dhowland/EasyAVR #1: Loading old layouts

cdarken, 2016-03-24:
Hi. I have my layout saved from an older version, namely: Easy_keymap_multiplatform_20150516 as the archive name says.

Was the backward compatibility lost at some point ?

dhowland, 2016-03-24:
There have been a few features that broke backward compatibility over the years.  Generally there are two ways this can happen.  First, when new features are added to the firmware or the keymapper, they may require a change to the save file format.  These can often be accommodated, but sometimes not, in which case it will warn you that the save file is too old.  The second possibility is that an update was made to a keyboard hardware description.  When this happens it always breaks backwards compatibility.  I try to avoid this, but the GH60 recently had one of those updates to add backlight support.

What keyboard are you using?

Your quickest way forward is probably just to remake your layout using the new tool.  Don't forget to use the copy/paste function to speed it up!

cdarken, 2016-03-24:
I have a QFR with Frosty Flake v2.
I will try to use the copy/paste, thanks.

closed 2016-04-02 by dhowland