dhowland/EasyAVR #7: Bouncing issue on Kitten Paw firmware

sagamusix, 2016-07-20:
I've been using a Filco Majestouch 2 with a Kitten Paw (rev 20130602) controller for about three months now, and recently some keys (most notably Return) started to bounce, i.e. being registered twice instead of just once. Using the original controller, or the costar_keyboard firmware by BathroomEpiphanies, this does not happen (but I'd like to keep my custom layout, of course :).
Please let me know if you need any further details.

sagamusix, 2016-07-21:
Note: I have set the debouncing time to 20ms which almost eliminates the problem (occasionally I am still getting double keys), but that seems very suspcious given that Bathroom Epiphanies told me that his firmware - which does not have the issue - has a debounce time of 6ms, while the hardware specs of the switches are around 5ms bounce time.

tmk, 2016-07-22:
The Bpiphany's controller has decorders or multiplexers to strobe matrix, unlike other simple plain keyboards. 
https://deskthority.net/w/images/7/71/Kitten_Paw_Schematics.pdf

I guess with this fact you need more time to settle signal on matrix before probing. EasyAVR delays just 5 cycles(312.5ns) by default, it seem to be enough for plain matrix driven by MCU I/O directly but not for the controller with decorders.
https://github.com/dhowland/EasyAVR/blob/master/firmware/src/matrix.c#L140
https://github.com/dhowland/EasyAVR/blob/de89ff278380d301396e37c84fa3369422f6a4d8/firmware/src/config_matrix.h#L35

You will have to count into decoder's propagation and transition time too, in addition to settling time of MCU I/O pin.
http://www.nxp.com/documents/data_sheet/74HC_HCT42_CNV.pdf

Meanwhile costar_keyboard uses 1us(1000ns) delay.
https://github.com/BathroomEpiphanies/costar_keyboard/blob/master/libraries/hw_interface.c#L45

I guess that 500ns is enough from datasheet but 1us is safer, of course. Try both 8(for 500ns) or 16(for 1us) for  `DEFAULT_MATRIX_SETUP_WAIT` and let me know your result.

dhowland, 2016-07-22:
You don't have to rebuild the software with a different DEFAULT MATRIX SETUP WAIT.  That value is configurable in the online config console and stored in EEPROM.  Try bumping the setup time.

There have been a few people in the GH support forum that have had similar problems with their bpiphany controllers.  The most recent that I dealt with did not ever find a fix, but when he get a replacement controller the problem disappeared.  I'm not saying it's the controller, because obviously the OG firmware works, but that it is a complex problem that could require some work.

If changing the setup time doesn't work, leave another message.  I have some ideas to try.

sagamusix, 2016-07-22:
Sadly, it doesn't seem like either values actually result in an improvement or change in behaviour.

dhowland, 2016-07-29:
I created a test build: https://www.dropbox.com/s/10bb3wri2m8o5na/test_build_sagamusix.zip?dl=0

It changes the USB rate.  let me know if this has any effect.

thanks.

sagamusix, 2016-07-29:
Sadly the problem appears to persist.