dhowland/EasyAVR #10: GH60 Rev. A

domoaligato, 2016-11-30:
I have a GH60 Revision A prototype that I am still using.

If i use the gh60 satan (rev b)  poker layout then 5,t,g,v, and space do not work.
If i use the gh60 (rev c) poker layout then 8,i,k, and m do not work.

hasu in tmk has used this code to support both boards. 
Could something like this be done in gh60_satan.py to support both pcb's?

static matrix_row_t read_cols(void)
{
    return (PINF&(1<<0) ? 0 : (1<<0)) |
           (PINF&(1<<1) ? 0 : (1<<1)) |
           (PINE&(1<<6) ? 0 : (1<<2)) |
           (PINC&(1<<7) ? 0 : (1<<3)) |
           (PINC&(1<<6) ? 0 : (1<<4)) |
           (PINB&(1<<6) ? 0 : (1<<5)) |
           (PIND&(1<<4) ? 0 : (1<<6)) |
           (PINB&(1<<1) ? 0 : (1<<7)) |
           ((PINB&(1<<0) && PINB&(1<<7)) ? 0 : (1<<8)) |     // Rev.A and B
           (PINB&(1<<5) ? 0 : (1<<9)) |
           (PINB&(1<<4) ? 0 : (1<<10)) |
           (PIND&(1<<7) ? 0 : (1<<11)) |
           (PIND&(1<<6) ? 0 : (1<<12)) |
           (PINB&(1<<3) ? 0 : (1<<13));
}

domoaligato, 2016-12-05:
there is more information on the tmk commit here. 
https://github.com/tmk/tmk_keyboard/commit/beacd5872c525b4bad215d2b2a4ff9ec60518e2f#diff-0377c7f68998f476a369c5fc487fc697

8carlosf, 2016-12-05:
you just need to change this line
https://github.com/dhowland/EasyAVR/blob/master/keymapper/easykeymap/boards/gh60.py#L43
to F0, F1, E6, C7, C6, B6, D4, B1, B0, B5, B4, D7, D6, B3 (given that tmk is correct, you just need to change the B7 to B0).

The led stuff will be wrong, but I don't think that is a problem.

I also recommend that you follow the custom boards instructions
https://github.com/dhowland/EasyAVR#supporting-custom-boards

Btw, I'm also new to this stuff, so I might be wrong..

domoaligato, 2016-12-05:
lol thanks. i will just create a new layout with all the correct info and submit a pull request.
I will also fix the led config.
I was just hoping someone might have some cool way of dynamically doing it like hasu did

closed 2016-12-05 by domoaligato