dhowland/EasyAVR #19: Power management key support

eltang, 2017-02-14:
There doesn't seem to be any way to map a key to be a Power key, a Sleep key, or a Wake key.

dhowland, 2017-02-14:
I need more detail. What scancodes specifically can't you send?

eltang, 2017-02-14:
[Here](https://en.wikipedia.org/wiki/Power_management_keys) is the Wikipedia article on them.

dhowland, 2017-02-14:
So EasyAVR currently supports the typical USB code pages for keyboard (0x07, 0x08), mouse (0x09), and the consumer page (0x0C) for MS-specific multimedia keys.  Those power keys are on the generic page (0x01).  A new HID interface is easy enough to add, but the problem is that the AVR is very short on RAM.  On the smaller devices, I probably don't have the space for an additional buffer.

However, I will give it a shot in a branch, see what happens.

eltang, 2017-02-15:
I believe it's possible to use one interface to report both media keys and power management keys.

dhowland, 2017-02-15:
According to the USB HID Usage Tables specification, section 4.5.1,

> Power controls can step the system through the following states: Full Power, Low Power, and Power Down. > Power control usages found in a System Control collection affect system level power. Those declared outside of a System Collection affect device level power.

We assume that "Sleep" and the other keys refer to the power state of the computer, not the keyboard, therefore I put it in a System control, not as a component in a keyboard.

Microsoft corroborates this in their HID driver docs by listing usage(0x80) as "System Controls (Power)"

eltang, 2017-02-15:
I'm not saying you should use the interface for reporting media keys as-is. I believe USB allows you to send different kinds of reports over the same interface at different times. See [this example](https://github.com/abcminiuser/lufa/blob/master/Demos/Device/ClassDriver/KeyboardMouseMultiReport/Descriptors.c#L48) [this example](https://github.com/kiibohd/controller/blob/133a98033e734dac99d8827db99acc75615aea0a/Output/pjrcUSB/arm/usb_desc.c#L270).

dhowland, 2017-02-16:
I got it to work in my test branch.  Now I have to just optimize everything and ensure I settle on a good implementation, then do some testing, but this feature will be available at some point.

dhowland, 2017-02-18:
Okay released a new version that has this feature.  Closing this for now.

closed 2017-02-18 by dhowland

eltang, 2017-02-18:
It looks like you made some changes to the media key support code as well. What was that about?

dhowland, 2017-02-18:
just some cleanup while I was in there.  I had only implemented parts of Microsoft's consumer page controls, this fills it out.

eltang, 2017-02-18:
Cool, thanks for taking care of this issue so quickly.