dhowland/EasyAVR #24: Feature request: separate option of multimedia keys and mouse pointer within config option

hschraep, 2017-03-10:
Hello, it is possible to separate within the "Configuration Options" the option to enable or disable the multimedia keys and mouse pointer for the "Toggle basic keyboard" function?

Keyboard acts as extended NKRO keyboard with mouse and media keys.
Keyboard acts as 6KRO keyboard plus mouse and media keys.

rgds hschraep

dhowland, 2017-03-10:
Okay I'm not sure I understand.  What mode do you want?  These are currently supported by the firmware:

- Keyboard is limited to standard 6KRO boot-compatible keyboard.
- Keyboard acts as 6KRO keyboard plus mouse and media keys.
- Keyboard acts as extended NKRO keyboard with mouse and media keys.

hschraep, 2017-03-11:
Hi i want Thiele mode:
Keyboard acts as extended NKRO keyboard with mouse and media keys

But without mouse.

dhowland, 2017-03-12:
I've made the changes in the Mouse_optional branch, but I haven't tested it yet

dhowland, 2017-03-12:
Well, that didn't work.

dhowland, 2017-03-13:
Okay this issue is resolved as of v2.05.03.

closed 2017-03-13 by dhowland

hschraep, 2017-03-13:
Hi checked it but if i disable the Mouse within NKRO keyboard, i have still 3 usb devices:
KezboardÖ NKRO, media-power kezsÖ ZES, mouseÖ NO.

within dmesg i have this:
```[ 6782.957218] usb 1-1.1: new full-speed USB device number 13 using ehci-pci
[ 6783.072810] usb 1-1.1: New USB device found, idVendor=03eb, idProduct=20ff
[ 6783.072813] usb 1-1.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 6783.072815] usb 1-1.1: Product: EasyAVR Multimedia Keyboard v2.05.03
[ 6783.076116] input: EasyAVR Multimedia Keyboard v2.05.03 as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:03EB:20FF.0035/input/input66
[ 6783.133669] hid-generic 0003:03EB:20FF.0035: input,hidraw2: USB HID v1.b0 Keyboard [EasyAVR Multimedia Keyboard v2.05.03] on usb-0000:00:1a.0-1.1/input0
[ 6783.136962] input: EasyAVR Multimedia Keyboard v2.05.03 as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:03EB:20FF.0036/input/input67
[ 6783.193369] hid-generic 0003:03EB:20FF.0036: input,hidraw5: USB HID v1.b0 Device [EasyAVR Multimedia Keyboard v2.05.03] on usb-0000:00:1a.0-1.1/input1
[ 6783.196138] input: EasyAVR Multimedia Keyboard v2.05.03 as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:03EB:20FF.0037/input/input68
[ 6783.253381] hid-generic 0003:03EB:20FF.0037: input,hidraw6: USB HID v1.b0 Device [EasyAVR Multimedia Keyboard v2.05.03] on usb-0000:00:1a.0-1.1/input2
```
I think it should only two devices one for the keyboard an one for the multimedia keys right?

If i connect my  cherry mx board 3.0 with multimedia keys i have only two devices:

```[ 6442.444460] usb 2-1.5.1: new low-speed USB device number 20 using ehci-pci
[ 6442.585451] usb 2-1.5.1: New USB device found, idVendor=046a, idProduct=010d
[ 6442.585453] usb 2-1.5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 6442.602728] input: HID 046a:010d as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.1/2-1.5.1:1.0/0003:046A:010D.0033/input/input64
[ 6442.662345] hid-generic 0003:046A:010D.0033: input,hidraw0: USB HID v1.11 Keyboard [HID 046a:010d] on usb-0000:00:1d.0-1.5.1/input0
[ 6442.674283] input: HID 046a:010d as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.1/2-1.5.1:1.1/0003:046A:010D.0034/input/input65
[ 6442.732575] hid-generic 0003:046A:010D.0034: input,hidraw1: USB HID v1.11 Device [HID 046a:010d] on usb-0000:00:1d.0-1.5.1/input1
```

dhowland, 2017-03-13:
No, by default EasyAVR has four devices.  Keyboard, media keys, power keys, and mouse.

hschraep, 2017-03-13:
But if i disable the mouse funktion it should be two, right? I have  it set to "KezboardÖ NKRO, media-power kezsÖ ZES, mouseÖ NO" but have three.

hschraep, 2017-03-13:
Ahh ok i understand now.