dhowland/EasyAVR #26: python error at reprogramming

hschraep, 2017-03-14:
Hello
i get this error if i do a build and reprogramming under ubuntu 16.04 with python 2.7 installed.
Is this a python problem? Something could not be found. But i could not figure out what.
dfu-programmer is installed under /usr/bin
and is working on the commandline to reprogram my keyboard.
But it would be nice if it works out of EasyAVR.

Here is the error:

[image]

dhowland, 2017-03-14:
Yeah I don't distribute dfu-programmer with the source code.  You have to install it yourself if you want to program directly through the keymapper.  Otherwise, you can just use it the old fashioned way: save a binary and program it with your tool of choice.

hschraep, 2017-03-14:
But it is installed:
`$ whereis dfu-programmer
dfu-programmer: /usr/bin/dfu-programmer /usr/share/man/man1/dfu-programmer.1.gz`

dhowland, 2017-03-14:
Oh, actually I think it's not constructing the command line correctly for linux.  Ugh, supporting windows and linux with one tool... not fun.  Especially when you can't test one of them.

8carlosf, 2017-03-14:
oh, I also had that problem, but assumed that it was my setup's fault. What I end up doing is generating the .hex file and using the `dfu-programmer` by and in the terminal (no big deal). I assumed it was because I need `sudo` to use `dfu-programmer` and I don't want to use `EasyAVR` with `sudo`, I don't think I even tried that.

dhowland, 2017-03-14:
pretty sure the problem is the quotes, which are for protection in Windows, but wouldn't be useful for *nix

hschraep, 2017-03-14:
Witch quotas you mean:
programming.py:
`loader_tools = [
        'dfu-programmer.exe',
        'dfu-programmer',
    ]`
or
 this:
`cmd = ('"%s" %s erase') % (self.tool_path, self.info.device.lower())`

dhowland, 2017-03-14:
I have committed a probable fix.  If you download the latest code from the Master branch you can try it

hschraep, 2017-03-14:
The double quotas i have also tryed
Same problem:
[image]

dhowland, 2017-03-14:
That screenshot is too small to read

dhowland, 2017-03-14:
Well then, you probably need to either fix permissions (chown) or make it executable (chmod +x)

hschraep, 2017-03-14:
I have updated the link.

hschraep, 2017-03-14:
Which file did you mean that i should set with chown or chmod?

dhowland, 2017-03-14:
/usr/bin/dfu-programmer

hschraep, 2017-03-15:
Hi,
dfu-programmer needs no chmod +x because it could be exec as normal user. So i have figured out that the ATm32U4DFU usb device is the problem why we need sudo to programm the keyboard.
Now i have defined a udev rule `sudo vim /etc/udev/rules.d/50-atmel-dfu.rules`
with inside:
```
# Atmel ATMega32U4
SUBSYSTEMS=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2ff4", MODE:="0666"
```
and now it is possible to programm the keyboard without sudo:
```
dfu-programmer atmega32u4 erase
dfu-programmer atmega32u4 flash EasyAVR/keymapper/my_gh60.hex
dfu-programmer atmega32u4 start
```

But we have still the error on EasyAVR:
[image]

hschraep, 2017-03-15:
I have try this to test if it works:
vim call.py:
```
#!/usr/bin/env python
import os
cmd = os.popen("dfu-programmer --help").readlines()
print cmd
```
this prints me with `python call.py`:
```
Usage: dfu-programmer target[:usb-bus,usb-addr] command [options] [global-options] [file|data]

global-options:
        --quiet
        --debug level    (level is an integer specifying level of detail)
        Global options can be used with any command and must come
        after the command and before any file or data value

commands:
        configure {BSB|SBV|SSB|EB|HSB} [--suppress-validation] data
        dump
        dump-eeprom
        dump-user
        erase [--suppress-validation]
        flash [--suppress-validation] [--suppress-bootloader-mem]
                     [--serial=hexdigits:offset] {file|STDIN}
        flash-eeprom [--suppress-validation]
                     [--serial=hexdigits:offset] {file|STDIN}
        flash-user   [--suppress-validation]
                     [--serial=hexdigits:offset] {file|STDIN}
        get     {bootloader-version|ID1|ID2|BSB|SBV|SSB|EB|
                 manufacturer|family|product-name|
                 product-revision|HSB}
        getfuse {LOCK|EPFL|BOOTPROT|BODLEVEL|BODHYST|
                 BODEN|ISP_BOD_EN|ISP_IO_COND_EN|
                 ISP_FORCE}
        setfuse {LOCK|EPFL|BOOTPROT|BODLEVEL|BODHYST|
                 BODEN|ISP_BOD_EN|ISP_IO_COND_EN|
                 ISP_FORCE} data
        setsecure
        reset
        start
[]
```

But im not a developer so i hope only this helps here to solve the problem. If i can try something for you i could do it.

dhowland, 2017-03-15:
Okay I re-read the python docs and found the problem.  I should be able to fix it pretty quick.

dhowland, 2017-03-15:
Alright, I've committed more changes to Master.  This should work.

The Popen object works somewhat differently on Unix platforms.  Thanks for doing the testing on Linux, I have no way to do it myself.

hschraep, 2017-03-15:
now it looks better but "launch" needs to be replaced by "start"
```
Usage: dfu-programmer target[:usb-bus,usb-addr] command [options] [global-options] [file|data]

global-options:
        --quiet
        --debug level    (level is an integer specifying level of detail)
        Global options can be used with any command and must come
        after the command and before any file or data value

commands:
        configure {BSB|SBV|SSB|EB|HSB} [--suppress-validation] data
        dump
        dump-eeprom
        dump-user
        erase [--suppress-validation]
        flash [--suppress-validation] [--suppress-bootloader-mem]
                     [--serial=hexdigits:offset] {file|STDIN}
        flash-eeprom [--suppress-validation]
                     [--serial=hexdigits:offset] {file|STDIN}
        flash-user   [--suppress-validation]
                     [--serial=hexdigits:offset] {file|STDIN}
        get     {bootloader-version|ID1|ID2|BSB|SBV|SSB|EB|
                 manufacturer|family|product-name|
                 product-revision|HSB}
        getfuse {LOCK|EPFL|BOOTPROT|BODLEVEL|BODHYST|
                 BODEN|ISP_BOD_EN|ISP_IO_COND_EN|
                 ISP_FORCE}
        setfuse {LOCK|EPFL|BOOTPROT|BODLEVEL|BODHYST|
                 BODEN|ISP_BOD_EN|ISP_IO_COND_EN|
                 ISP_FORCE} data
        setsecure
        reset
        start
[]
```

dhowland, 2017-03-15:
What version are you using?

```
D:\Code\EasyAVR\keymapper\easykeymap\exttools>dfu-programmer.exe --help
dfu-programmer 0.7.0
https://github.com/dfu-programmer/dfu-programmer
Usage: dfu-programmer target[:usb-bus,usb-addr] command [options] [global-options] [file|data]

global-options:
        --quiet
        --debug level    (level is an integer specifying level of detail)
        Global options can be used with any command and must come
        after the command and before any file or data value

command summary:
        launch       [--no-reset]
        read         [--force] [--bin] [(flash)|--user|--eeprom]
        erase        [--force] [--suppress-validation]
        flash        [--force] [(flash)|--user|--eeprom]
                     [--suppress-validation]
                     [--suppress-bootloader-mem]
                     [--serial=hexdigits:offset] {file|STDIN}
        setsecure
        configure {BSB|SBV|SSB|EB|HSB} [--suppress-validation] data
        get     {bootloader-version|ID1|ID2|BSB|SBV|SSB|EB|
                 manufacturer|family|product-name|
                 product-revision|HSB}
        getfuse {LOCK|EPFL|BOOTPROT|BODLEVEL|BODHYST|
                 BODEN|ISP_BOD_EN|ISP_IO_COND_EN|
                 ISP_FORCE}
        setfuse {LOCK|EPFL|BOOTPROT|BODLEVEL|BODHYST|
                 BODEN|ISP_BOD_EN|ISP_IO_COND_EN|
                 ISP_FORCE} data

additional details:
 launch: Launch from the bootloader into the main program using a watchdog
         reset.  To jump directly into the main program use --no-reset.
   read: Read the program memory in flash and output non-blank pages in ihex
         format.  Use --force to output the entire memory and --bin for binary
         output.  User page and eeprom are selected using --user and --eprom
  erase: Erase memory contents if the chip is not blank or always with --force
  flash: Flash a program onto device flash memory.  EEPROM and user page are
         selected using --eeprom|--user flags. Use --force to ignore warning
         when data exists in target memory region.  Bootloader configuration
         uses last 4 to 8 bytes of user page, --force always required here.
Note: version 0.6.1 commands still supported.

```

hschraep, 2017-03-15:
This is the standard version on ubuntue 16.04 TLS:
```
~$ dfu-programmer --version
dfu-programmer 0.6.1
```

dhowland, 2017-03-15:
WTF?  Goddamn package managers over there are slacking.  Version 0.7.0 is almost 3 years old

dhowland, 2017-03-15:
okay i've committed what should be the last fix.  I've tried it on Windows and it works

hschraep, 2017-03-15:
Greate job now it works. If you have something to test within linux you can ask me ;-). 
[image]

dhowland, 2017-03-15:
Great, thanks.

closed 2017-03-15 by dhowland