dhowland/EasyAVR #27: Media Controls on GH60 doing nothing on the current version. Worked on the now 10 month old version.

Gregorovich, 2017-03-26:
I haven't done much testing, but this appears to be the case.

Media controls including volume and media play/pause, next track etc. are not working under the current version. It was working on the 10 month old release. I wouldn't know where to start looking for how to solve this in the code just yet.

dhowland, 2017-03-26:
Hmm, I thought this stuff was working correctly because I tested while I was just refactoring it.

Can you try making a brand new layout for your GH60? (File->New...)

If that works, then the problem is with loading old file formats.  If it still doesn't work, we'll have to do some more intense debugging.

Gregorovich, 2017-03-27:
I haven't tried any of the different versions between 60f26ea and the current version of 41a64ad, but by default (without changing any console settings), volume and media controls aren't working.

I did originally have to create a new layout in any case as the old .dat file for 60* would not import into the 41* version, but I just checked out the old version and ran both simultaneously to manually import all 10 layers  : ) .

dhowland, 2017-03-27:
What OS are you using?

Gregorovich, 2017-03-27:
I am running Windows 8.1 64bit. I have a Linux system that I haven't yet tried 1.

dhowland, 2017-03-29:
I just programmed my Epsilon with the latest firmware and the media keys are working fine for me.  Calc, and volume, etc.

Can you open the "Devices and Printers" control panel, right-click your Easy AVR device, select Properties, and choose the hardware tab.  Then take a screenshot.  It should look something like this:
[image]

Gregorovich, 2017-03-29:
With the old version:
[image]

With the new version:
[image]
with another USB Input Device below it.

Apparently there's a problem with the driver for the HID-compliant mouse - but the left and right clicks are working just fine.

I am flashing it using the tkg-toolkit, but I doubt that would affect anything...

dhowland, 2017-03-29:
The important bit is the consumer control device, which seems to be working.  But if you have any broken endpoints, then something is definitely wrong.  This is very hard to debug if I don't have an example in front of me.

Let's start with some basics.  Can you flash it to the new version, and then plug it into a different computer?  I want to remove as many variables as possible, that means ensuring that the problem follows the keyboard.

Gregorovich, 2017-03-29:
Well this is weird. My guess is that this was in fact just something wrong during re-initialisation after writing the firmware. Unplugging it and plugging it back in appears to have worked - (I plugged it into my laptop, and it was working there, plugged it back into my desktop and it started working with the volume and media keys).

There are no longer any orange triangles any more in the properties - and judging by the icon and what you've said, and that there are now 2 consumer control devices, which I would imagine may well have mouse functionality, might explain why it wasn't working fully, including media functionality.
[image]

As to why it doesn't work without replugging it, and it used to, I don't know. This isn't much of an issue, and may well be related to the fact that I'm using tkg-toolkit to flash the firmware.

dhowland, 2017-03-29:
The reason is that between the old version and the new version, I changed the USB descriptors around.  So, it appears that windows re-enumerates when it comes out of reset, but won't actually replaces the devices unless there is a plugin event.

closed 2017-03-29 by dhowland