dhowland/EasyAVR #31: Feature requests; scroll up/down, rapid fire available for media keys, accessing the default layer from another layer, clipboard.

Gregorovich, 2017-04-21:
1. Mouse scroll up/down buttons.
There are mouse buttons 3, 4 and 5, but no scroll (unless that's what 4 and 5 are, but I wasn't expecting them to be. I could test this, but I have other feature requests as well so I may as well ask : ) ) - Ideally I'd like these to be rapid fire-able.

2. Rapid fire mode available for Media Keys.
I use the media keys, especially the volume up/down buttons. As such, I would like these to have the option to be rapid fire, as currently there is no option for this and I believe it could be possible. I use rapid fire on the keys I have mapped to backspace, delete, and arrow keys (48 ms min hold time (default is 800), default / 32 ms repeat delay), and as such would like it on my volume control too as I miss not having it. This may also be nice for macros, but that may be hard to add, as it would conflict with the length of the macros, so I don't know if this would be doable.

3. Accessing the default layer from another layer.
Default layer / layer 0. If I'm on another function layer, would there be a way to get back to the default layer? For example, If I was to have the "Caps" key mapped to "Fn", "W" (Qwerty layout) mapped to "Ctrl", and then I want maybe the "Enter" key mapped to "Fn0", such that I can do something like Ctrl+C without moving my hands from the home row or moving them to somewhere uncomfortable as the bottom left of my keyboard / "LCtrl" is uncomfortable to type. Whilst this is a very specific example, I'm sure there could be other uses for this. I know that the default layer can be changed, but then how would you get to layer 0?

4. Clipboard.
A temporary copy / paste layer, or a clipboard-style history thing. If (like me) you are using all 10 layers, and then decide you want to move some layers around, whilst the copy / paste function is nice, it isn't as helpful as it could be. Of course, the layers don't necessarily need to be moved around, but it makes it slightly easier to customise the layout if the similar layers are "grouped" (at least in my mind).

Finally, I would like to thank everyone for all their work so far on this project as it makes my one-of-a-kind keyboard possible : )

P.S.: I haven't used Git much, so I'm hoping the "Issues" page is the right place to put this : )

dhowland, 2017-04-22:
Yeah, the issues page is the correct place for this, although it's a good idea to put separate feature requests into separate issues.  This allows each to be tracked individually.

1. Mouse scroll up/down buttons.
I considered this once or twice, but I decided against it for a couple reasons.  One is the complexity of supporting that kind of function, but mostly because there seems to be no point to using it on a keyboard.  You have arrow keys and pgup/pgdn for scrolling around.

2. Rapid fire mode available for Media Keys.
In the world of USB HID, not all functions are the same.  They are several modes which are supposed to conform to the way that humans interact with them.  It means dealing with actions relatively or absolutely, allowing holds or making it instantaneous, etc.  EasyAVR doesn't really have enough time to get complex about this, so it just sends the scancodes and hopes for the best.  This is why I didn't add those advanced functions to Media keys.  Perhaps this request could work, I'm not sure.

3. Accessing the default layer from another layer.
Yeah, I guess there isn't a key for accessing the default layer.  However, you don't really NEED it.  You can just copy your entire default layer and paste it to L9 or whatever.  Then you can use Fn9 for the same thing.

4. Clipboard.
I don't understand what you're asking for here.  Copy/Paste is a clipboard. You can think of it as temporary.

So, in conclusion, I probably won't be able to find the time to get to 1,2.  3 is interesting in that there should probably be a Fn0 for completeness, but I'm not sure I'll be able to get to it right away.  4 I don't think so.  If you really need some of these changes, you may want to fork the code and add it yourself, then submit a pull request.

Gregorovich, 2017-04-26:
We could close this and I could open 4 individual issues...

1. In which case, I will look into scroll up/down as in a terminal (when not inside of screen / tmux) I prefer using the scroll wheel as opposed to moving the whole page up / down. I won't however have time until early June, unfortunately.

2. That makes sense. I wouldn't mind looking into this as well.

3. True. As I don't really do this, it's not something I'm looking for, so probably won't look into.

4. It is, however as I do actually use all 10 different layers, should I want to rearrange them, I can copy one of them into the clipboard, but pasting it on a new layer would result in the loss of one layer. Yes, I could just screenshot it, but it's a nice feature to have, but I can live without it : ) .

So I will look into 1 and 2 early June when I have time, if noone looks into this before then. 3 and 4 were things I thought would be nice to have, but I probably won't look into them.

dhowland, 2018-03-04:
This is an old issue.  #3 was recently added to the firmware.  The rest of it was addressed in discussion.

closed 2018-03-04 by dhowland