dhowland/EasyAVR #33: Feature request: Apple key codes

punkpc, 2017-04-27:
Requesting Apple OSX specific key codes.

Like listed here...
http://eastmanreference.com/complete-list-of-applescript-key-codes/

A lot of the Windows commands work with Mac, but there are a few that are OSX specific.
Like the Mac function keys. 

Brightness down (Monitor/Screen)

Brightness up (Monitor/Screen)

Mission Control

Launchpad

Media Eject

Thanks

dhowland, 2017-04-27:
EasyAVR implements a USB HID keyboard, which sends scancodes based on HID code page 7.  As far as I know, it implements all keys from that page that are used in both Windows and Mac.  If there are HID codes that should be supported but aren't, let me know which ones specifically.  The link you provided doesn't reference any USB HID scancodes.

punkpc, 2017-04-27:
Oh ok. I will see if I can find the scancodes for those commands.

bmmcwhirt, 2017-04-30:
Brightness is already implemented as SCANCODE_BRIGHT_DEC and SCANCODE_BRIGHT_INC. Or most Macs will be configured to do this on F14 and F15 which can be accessed with HID_KEYBOARD_SC_F14 and HID_KEYBOARD_SC_F15.

Eject I am not sure about

Mission Control and Launchpad are shortcuts.
Mission control by default in Sierra is CTRL UP_ARROW and CTRL DOWN_ARROW to exit. Likewise CTRL LEFT_ARROW and CTRL RIGHT_ARROW move left and right a space respectively.

On my machine Launchpad is not mapped to anything.

These are all in Preferences in Keyboard on the Shortcuts tab.

Hopefully this helps, if you have further issues replay and I will try to help.