dhowland/EasyAVR #34: TADA68 implementation doesn't work on Saber68

Zinglish, 2017-04-27:
- Create a new TADA68 layout.
- Save as FLASH.BIN.
- Copy the file to flash to keyboard storage.
- Press ESC on keyboard.

Keyboard is completely unresponsive. I had to reload the backup I had of the layout to get the keyboard working again.

Any ideas?

dhowland, 2017-04-27:
I don't know what a Saber68 is, but I suspect that EasyAVR isn't compatible with the USB mass storage bootloader.  I do not own a TADA68 and I've never tested it myself.

Zinglish, 2017-04-27:
Saber68 is basically a re-branded TADA68 I guess. So perhaps the alternative question is probably whether the TADA68 implementation works on that keyboard.

dhowland, 2017-04-27:
[I believe the user who contributed it had to flash the classic DFU bootloader to get it to work.](https://www.reddit.com/r/MechanicalKeyboards/comments/66sji0/bricked_tada68_and_how_i_fixed_it_a_novices_tale/)

If there is documentation for the Saber bootloader I might be able to get it working, but I really don't know what they have on that thing or who wrote it.

Zinglish, 2017-04-27:
Don't worry about it 👍 I managed to flash a new layout to my Saber using the TADA68 Chinese tools available.

I do believe he must have flashed his TADA68 with that bootloader to get EasyAVR working as expected.

Thanks for your time!

closed 2017-04-27 by Zinglish