dhowland/EasyAVR #47: Changing the breathing effect speed

raymond-liang, 2017-10-31:
Which files would I need to modify before compiling and how would I generate a new build? Specifically for a GH60 Satan.

I am a programmer, but I don't have experience with programming microcontrollers.

dhowland, 2017-10-31:
The breathing function on EasyAVR is very primitive.  As far as I remember, the brightness changes are a linear sweep and therefore the breathing is actually a triangle wave.  I'm almost positive that what you want is actually to implement a proper sine wave.  But before we really talk about that, I need to discuss the main problem.

When I started the firmware, it only supported my Epsilon project.  Later, I ported it to many other boards.  In order to use a common interface for keymapping and programming, the firmware has to be pretty much the same on all the different hardware.  So there are many boards that have LEDs hooked up to the AVR in ways that prevent it from using the AVR's hardware PWM.  Therefore, I had to implement software PWM, which is much slower.  A consequence of this is that when it is very dim, you can actually see the pulses.  If you look closely at the breathing mode, you can probably spot it.  This is one reason why I made the breathing so fast.  It spends very little time at the lowest brightness.  Still, through some clever coding, I was able to minimize this effect.  It stresses the AVR, though.  It is not a powerful processor, so I have to be very "thrifty" with my code.

So, given all of that, in order to have a more pleasing breathing I think what we need to do is to give it a map for a real sine wave (computing it real time would be needlessly expensive).  The code would have to be written to time it out properly, then that would have to be integrated into the backlight code.  This is a fairly complicated task.

To recompile, you need Atmel Studio to open the project.  Then, there are compile scripts in the root directory of the project that use Atmel Studio to recompile all of the binary templates.

This is an interesting issue, I'll look into it more.

dhowland, 2017-11-02:
Okay so I looked into it more, and refreshed my memory as to how the LED code works.

I changed BACKLIGHT_CHANGE_DELAY to (96) in config_led.h and recompiled all binaries with incorporate.bat.

The breathing speed is slower by half.  However, the software PWM only has 16 levels of brightness.  This leads to the change in levels being easily visible while it is moving in and out.  It doesn't look very good IMHO.  I have saved the .hex files if you would like to try it yourself.

raymond-liang, 2017-11-02:
I would like to try it out yes, thank you for looking into it.

I meant to reply to your response but have been too preoccupied with other things. In my case, do you know whether the GH60 Satan has the LEDs attached to the AVR in such a way that the PWM cannot be used? If the hardware PWM could be used to achieve smoother breathing, would it not be simpler to implement a solution that used the hardware PWM? Could EasyAVR be changed to then enable/disable hardware PWM based on whether a certain board supported it?

dhowland, 2017-11-02:
Here's a link to the built binaries. (.hex files)
https://www.dropbox.com/s/438d9xio8p6pafm/slow_breathe_builds.zip?dl=0

You can replace them in your installation of the keymapper and then create a new build for your Satan, and it will be slower.

It is possible to add HW PWM to EasyAVR, yes.  It would require new code, but it would be simpler code than the software PWM.  I don't have time to look into it now.  

As for the Satan, it looks like it drives backlights off PORTB, so HW PWM should work.