dhowland/EasyAVR #48: Macro parsing Shift seems to be off by one

gorauskas, 2017-12-23:
I have a custom board built on the GH60 Satan sold by 1Up keyboards. I created a macro with the following text:

    jgorauskas@gmail.com

I compile and flash the board ... then I go to execute the macro and I get this output:

    jgorauskas2Gmail.com

I have other examples too ... It seems that the macro parsing / emitting of the codes is off by one when it sees a need to use shift ... at least on the GH60 Satan that I have. I have not tried any other PCB.

Let me know how I can provide any additional information to help debug this issue.

dhowland, 2017-12-23:
I just tried it on my Zeta with the latest firmware and it worked perfectly.  So... this makes your issue very hard to diagnose.

I'm afraid in order to get to the bottom of this, you're going to have to do some testing.

- What mode is your board in?  NKRO or 6KRO?
- Have you tried a fresh new config from scratch?
- Does the same thing happen when you plug the keyboard into a different computer?
- Are you using any macro tools on your computer, like autohotkey?
- What happens when the macro has lots of symbols or capital letters in a row?

If all these things fail, get EK switch hitter, hit the macro while in the tool, and send the log file of the output.

gorauskas, 2017-12-26:
@dhowland  I dug a little deeper as you suggested and I found that the firmware is working as expected in Windows 10 and in MacOS Sierra. I, also, am unable to repro the issue on those systems. Those are the other systems I have here at home. 

However, as originally described, the issue persists on my main machine which is running Arch Linux with kernel 4.9.65-1-lts on a XFCE 4.12 desktop. Here are answers to your questions:

1. I am using 6KRO ... I used console mode to toggle that
2. Not yet, but will try that next
3. No ... it works on Windows 10 and MacOS Sierra
4. Not using Autohotkey on my main Linux system. But I am using a tool called Synergy (https://symless.com/) to share keyboard and mouse across different systems. Both systems are running the same version of Arch linux as described above. When I turn Synergy off, the problem still persists.

For your last bullet point ... I tried the following text in a macro:

    AsDfGh!2#4%6<.?[} QWERTY

and I get the following output when I run it:

    aSdFgH1@3$5^<.?{] qWERTY

More later ...

gorauskas, 2017-12-26:
In linux systems using X11 there is a tool call `xev` that logs all events that X captures. I created a gist from that capture here: https://gist.github.com/gorauskas/da9024f4fbd63397dfccabdd0bf396e5

The macro I used during the capture writes out the following text:

    AsDfGh!2#4%6<.?[} QWERTY

dhowland, 2017-12-26:
Linux compatibility strikes again! Thanks I know what is happening. That log will be perfect.

Clearly Linux processes the modifiers after the alphas, opposite of Windows. 

I'll need to look at the macro logic when I get home.

gorauskas, 2017-12-26:
Here is another output of xev ... this time is the full log, as I previously trimmed the output that was not relevant to the macro. The macro execution starts at line 94 in this log. prior to that line, it displays window setup events by X. It should print my email address, like so:

    jgorauskas@gmail.com

https://gist.github.com/gorauskas/dafb792870b6ec08d376c0452f069d3c

dhowland, 2017-12-27:
I have committed an update that will hopefully fix the issue.  Can you try it?

gorauskas, 2017-12-27:
I will try it a little later today on PST time.

gorauskas, 2018-01-04:
@dhowland  I can confirm that the fix is indeed good and it works as expected in Linux now.  Thank you for the quick work on this!

dhowland, 2018-01-04:
No problem, thanks for the follow-up.

closed 2018-01-04 by dhowland