dhowland/EasyAVR #50: Request: two key support

pimaxc, 2018-01-02:
https://mehkee.com/products/macropad-2-switch-pcb?variant=45125555663

dhowland, 2018-01-06:
Do you have a pinout diagram or anything else that describes the matrix?

pimaxc, 2018-01-06:
Switch pins are 2 and 3 source code for their arduino sketch is here. https://github.com/di0ib/Misc/blob/master/2pack/2pack.ino

tahuds, 2018-01-17:
This would be awesome.

dhowland, 2018-01-17:
This is going to be the last board I do by myself.  Going forward I'm going to require that people at least try to follow the "Supporting custom boards" section of the README.

I have added initial support for the 2key but it requires testing.  Please give it a shot and let me know if it works.  Thanks

tahuds, 2018-01-18:
Thank you!  I'm guessing from your picture you don't have a lot of free time and EasyKeyMapper has made a lot of people's lives better. Sorry for bothering you.

dhowland, 2018-01-18:
No need to be sorry, it's cool.  What would really help is some testing.  I don't own the hardware, so I'm not able to do it myself.  Thanks

tahuds, 2018-01-18:
I've been waiting on getting the pro micro board for the pcb. Once they are
in stock I'll let you know! :)

pimaxc, 2018-01-29:
I'm trying but looks like I need to flash a new boot loader to get it to recognize and I don't have the hardware for that right now.

dhowland, 2018-01-29:
okay.  too bad.

closed 2018-01-29 by dhowland