dhowland/EasyAVR #51: SyntaxError

remlap, 2018-01-06:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/remlap/Desktop/EasyAVR/keymapper/easykeymap/gui.py", line 61, in <module>
    import easykeymap.newboard as newboard
  File "easykeymap/newboard.py", line 43, in <module>
    import easykeymap.kleparse as kleparse
  File "easykeymap/kleparse.py", line 58
SyntaxError: Non-ASCII character '\xc2' in file easykeymap/kleparse.py on line 58, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
```

dhowland, 2018-01-06:
Hi, thanks for the bug report!

Python 2.7 is no longer supported.  Now that it is 2018 I only test on Python 3.6.  (there has been a notice on the README for year).

However, if all it takes is an encoding string to fix this issue, I can add that.  It doesn't hurt anything.

remlap, 2018-01-06:
I changed the easykeymap.sh to python3 (I have 3.6) and now works fine.

Thanks.

dhowland, 2018-01-06:
Great.  Also the encoding string has been added to source.
thanks

closed 2018-01-06 by dhowland

remlap, 2018-01-06:
That also works, thanks for everything!