dhowland/EasyAVR #52: Why is there no left alt support for keys with mods

slugger7, 2018-01-22:
Hi All this might not be an issue I was just wondering why there is no support for assigning a key with left alt.
[image]
Every key that I want to have mods pressed down with that have left alt I have to create a macro for

dhowland, 2018-01-22:
AVR microcontrollers are very resource constrained, and the implementation of the firmware is influenced is every way by these constraints.  Because of memory conservation efforts, there are only 4 bits available to store the mod flags.  This is usually just fine because if you want Shift+a to make an 'A', it doesn't matter which shift key you hold.  The exception is Alt, because users of foreign layouts usually require AltGraph to make their umlauts and accents and whatever else they need, and that only works on the Right Alt key.  I have never run into a situation where a key combination only works with L_Alt.

dhowland, 2018-01-22:
By the way, all four of the mods are the "right" versions.  Alt is the only one that indicates as much because of a user-submitted a patch.  He wanted to make it obvious to international users that AltGr would work.

slugger7, 2018-01-23:
Interesting. Thanks for the info. The reason I asked this is because in Linux mint l_alt + F2 lets you run a command in the background but for some reason r_alt + F2 does not let you do this. Now I know I can go into the os and bind it to the right alt variant. It was just odd  for me.

closed 2018-01-23 by slugger7

dhowland, 2018-01-24:
Hmm, that's interesting.  I did not know that kind of thing happened in Linux.  As far as I knew, LAlt+F2 would be expected to do the same thing as RAlt+F2.

On Windows I've found exceptions in two places.  One in VirtualBox, which uses RCtrl as a "host" key.  I figured they did so because of the same line of thinking.  Since both Ctrl keys do the same thing, they can take one and nothing of value is lost to the guest OS.

The other is video games.