dhowland/EasyAVR #53: .dat files wont load

remlap, 2018-02-02:
Hi recently switched to Ubuntu Budgie and tried to load some of my .dat files and EasyAVR keeps giving me:

"The save file describes a configuration that cannot be loaded."

What can I do to get my layout dat file to be compatible?

dhowland, 2018-02-03:
What board are you using?

That error message is shown when the hardware description of a board is updated.  The layout saves are tied to switch matrix of the hardware, so if it changes and you try to load an old file, it will produce a broken build.  Hardware descriptions don't change very often.

Sounds like you also upgraded to a newer version of EasyAVR when you upgraded your OS.

remlap, 2018-02-03:
GH60 Satan is the board but I've just built a new layout and flashed it.
Thanks for the update.

dhowland, 2018-02-03:
Alright, yes, the Satan was updated when a user found a bug on the split right shift.  If you can't remember what was in your old layout you should be able to open for viewing using an old version of the tool.  Sorry for the trouble.

closed 2018-02-03 by dhowland