dhowland/EasyAVR #54: Increased number of macros/layers ?

Synless, 2018-02-10:
Hi !
First thanks for Easy AVR, I just modded a Filco TKL with a Pegasus Hoof and it does wonder.

I wanted to type characters like  _ő è ú í ü_  and others unicode (French/German) but currently it is not possible from a US layout (Exception unexpected escape sentence when I try to build with a unicode). I wish to keep the US payout but be able to print special characters.
I have seen that [it was discussed already](https://github.com/dhowland/EasyAVR/issues/35), and I though of a workaround : using Alt+XXXX to print them. But since a lot of macros is required I try to increase the maximum number of macros and layers.

(Disclaimer, I am not great with Python).
I went to guy.py and changed the number of macros from 14 to 16 as a beginning and then I added M15, M16 while pushing aside some other macros to get some bytes available (from the second parameter of the scancode table).
https://github.com/dhowland/EasyAVR/blob/8a390dbe5475d006a2af9c3107d80a1c2ba10033/keymapper/easykeymap/gui.py#L85
https://github.com/dhowland/EasyAVR/blob/8a390dbe5475d006a2af9c3107d80a1c2ba10033/keymapper/easykeymap/scancodes.py#L304
https://github.com/dhowland/EasyAVR/blob/e3ecba61278e55843f3f20b1ec3c8b4ced51d3fe/keymapper/easykeymap/picker.py#L318
But then if I try to select a macro the log window on the back displays an exception :
`File "C:\Users\Synless\Desktop\EasyAVR\keymapper\easykeymap\gui.py", line 463, in selectmacro self.macros[self.currentmacro] = macro_string`
IndexError: list assignment index out of range
At this point I am not able to debug and play around with the code so I ask for help to increase the maximum number of macros (and layers ?).

Any help provided would be appreciated. 
And once again, great piece of software.

dhowland, 2018-02-11:
Adding more macros to the firmware is more complicated than you think.  It requires a change to the C code.

Synless, 2018-02-11:
Could you give hints on where to look for ?
Would it be possible to have more macros and layers in the foreseeable future ?
If not, could you think to another solution for unicode on custom controllers ?

dhowland, 2018-02-11:
Sorry, I was taking care of a baby when I responded.  I committed a change that demonstrates how to increase the number of macros.  If you want to make your own branch with even *more* macros, you can use the latest code as a base.

Unicode is really an issue for the PC, rather than the keyboard.  They keyboard sends scancodes, not characters.  The PC is responsible for translating the scancodes into characters, and in languages with many diacritics/umlauts/accents/whatever, there are more characters than there are keys.  If you need to enter those characters, you might want to look into internationalization.

closed 2018-02-11 by dhowland