dhowland/EasyAVR #56: Remove password generator feature

dhowland, 2018-03-02:
The password generator function is flawed.  It's not that useful for entering passwords because it doesn't travel with the user.  It's hard to use because it requires entering a perfect string without visual aids to get the right password.  Also it doesn't generate characters with a uniform distribution.

I don't think anyone uses it.  I sure don't.

The function started as an investigation into a concept of a keyboard that changes what each key outputs each time it is pressed.  Think of the enigma machine.  What I ended up with didn't really resemble that at all.  Perhaps I will replace it with something like that just for fun.  Haven't decided yet.

In any case, the password function should be removed.  It takes up a bunch of space I'd rather have back.

gorauskas, 2018-03-02:
+ 1 

Use that space for more macros  ;)

closed 2018-03-19 by dhowland

nodswal, 2019-03-07:
I liked this feature, I was planning getting a second keyboard and other wise its in lastpass, but a majority of the time it would be nice to just type the password.