dhowland/EasyAVR #57: Reimplement the keymapper GUI

dhowland, 2018-03-02:
The code for the keymapper is a mess.  It grew organically from a simple utility to a complex application.  There are several problems.  The code is not functionally decomposed into logical sections (spaghetti code).  The save file format is finicky and hard to use.  It looks awful because of TK.  It makes huge exe builds because of TK.

A new keymapper should be built from the ground up with wxPython.

gorauskas, 2018-03-02:
+1

Just curious... why wxPython?

dhowland, 2018-04-06:
Going to call this one done.  There appears to be an issue somewhere that causes wx to crash Python, but it isn't repeatable.  Other than that I think this is done.

closed 2018-04-06 by dhowland