dhowland/EasyAVR #60: Fails to build binary

Cabalforces, 2018-03-11:
[Throws this error message](https://imgur.com/a/zKex7)

dhowland, 2018-03-11:
What is the hardware and layout?  Can you provide the save file?
thanks.

Cabalforces, 2018-03-11:
It's a teensy, the layout is a numpad with an extra row
[numpad.zip](https://github.com/dhowland/EasyAVR/files/1800791/numpad.zip)

dhowland, 2018-03-12:
The problem is the definition of the board.  You have this:

```python
num_rows = 6
num_cols = 4
```
And then later you have this:

```python
((4, 4), (0, 6), 'SCANCODE_FN0')],
```
That assigns the key to matrix row index 0 and column index 6.  (that's the 7th column)  There are only 4.

The matrix assignments for all the keys require fixing.

Cabalforces, 2018-03-12:
Thank you very much for your help, it was driving me crazy. I've read through the whole code at least ten times, and I didn't find the mistake.

dhowland, 2018-03-12:
No problem, and I commend you for sticking with it.  Because the tool has to support so many different kinds of boards, these files end up being very complicated and few people have managed to understand them.

But you have reminded me that I really need to put a sanity check into the config loader so these common issues are pointed out immediately when you load it.  I'll add that to 3.0.

dhowland, 2018-04-05:
Quoting from the Geekhack thread:

"if you try and build a COSTAR sized matrix against the FULLSIZE firmware it doesn't work well"

In addition, Pull Request #37 tried to fix this kind of problem a year ago.

Two more data points that show this Issue needs to be fixed.

I'll be adding some defensive code soon.

closed 2018-04-06 by dhowland