dhowland/EasyAVR #65: French symbols not usable

Alkeryn, 2018-03-28:
So i can't set altgr or éèçà and such others
even when pushing on my french keyboard the software don't recognize them ^^

Synless, 2018-03-29:
You have to use the dead key on a english international keyboard for that (windows keyboard settings).
ç does not exist and is not supported , like all other unicode : (

dhowland, 2018-03-29:
EasyAVR does not have internationalization support.  Keep in mind that keyboards don't, either.  It's the PC that interprets the special characters.  No matter which kind of keyboard you have, the key to the right of the spacebar sends the same scancode, no matter if it is labeled "Alt" or "AltGr".

Alkeryn, 2018-03-30:
What i mean is that qmk have unicode character suport
and i'm not using windows ^^

dhowland, 2018-04-02:
Duplicate of #35

closed 2018-04-02 by dhowland