dhowland/EasyAVR #67: Implement scaling for high-DPI monitors

dhowland, 2018-04-02:
The current GUI includes several hard-coded pixel sizes that will not look right on a 4k monitor.

dhowland, 2018-04-07:
I tried to test 4k today by using Nvidia DSR to double my screen resolution.  It didn't work.  I have no way to test this, so the Issue is going to have to wait. :(

gorauskas, 2018-04-14:
This is what it looks like on a Seiki 4K tv at 30 Hz that I use as a monitor.  DPI is set to 96.

[image]

Does this help in anyway? ... or are you testing for something different?

dhowland, 2018-04-16:
Wow this is really great feedback.  Thank you very much.

The GUI is written in such a way that a 1u key is 48x48 pixels.  That looks good on my 1080p screen, but the assumption was that on a 4K screen the buttons would be too small.  On a high DPI screen the text is usually scaled to remain readable, thus the size of the buttons should also be scaled.

The screenshots you've provided don't show that, but you also aren't using a high-DPI screen.  Regular HD monitors are usually around 100 DPI and 4K are usually around 200 DPI.  So your TV is so large that the DPI is actually about the same as an HD monitor, so I still need more research.

This whole thing is complicated because the OS can usually perform automatic DPI scaling.  I don't know how that would affect the tool.

Oh well.  If there is a problem, eventually someone will show up complaining about the interface, and I can fix it then.