dhowland/EasyAVR #71: Turn Backlight off when the computer is in standby mode

cl0rm, 2018-04-10:
thanks very much for this great firmware, it worked great with my Gigabyte Osmium and a teensy 2.0++.

However, there is one big flaw so far. The backlight stays on when the usb is powered, and it does not detect a when the computer is in standby mode.

This is possible, as many other input devices do that (including the original controller on my Gigabyte Osmium), and there is also a fix for QMK to support it. As far as I remember there is a standby event built into the USB/HID interface. ('USB suspend').
In the suspend mode, the keyboard is still responsive so it can wake the computer, but the backlight is turned off.
It would be very nice to implement that into EasyAVR as well, as it is quite basic.

Maybe it is even possible to set only specific keys to wake the PC, but I don't know how hard it is to code that, but it would be a very nice and useful feature, so that you do not accidentally wake the PC.

dhowland, 2018-04-10:
EasyAVR already detects USB Suspend, because there is an LED indicator for it.  Shouldn't be very difficult to disable backlighting.  Of course, probably not everyone wants this.  For example, some people program the keyboards to "breathe" when the PC is asleep.  However, I agree it should turn off.  Many custom keyboards draw tons of power for the backlights, and many homemade boards draw considerably more than the USB spec allows.  I'll give this a shot.

cl0rm, 2018-04-10:
Thanks...
Maybe you can add an option for setting a separate lighting mode for standby.
Then the user can decide which one suits the best.

dhowland, 2018-04-12:
I added this and it works.  Going into the next version.