dhowland/EasyAVR #72: Boot Key for Teensy

cl0rm, 2018-04-10:
Is there a way to get the boot key working with the PJRC teensy bootloader?

When I press it, the teensy does indeed reboot, but not into the teensy bootloader, it seems like another boot mode which does not work correctly.

Tested with a teensy 2.0++

That might help for this:
https://www.pjrc.com/teensy/jump_to_bootloader.html

dhowland, 2018-04-10:
It should be implemented to work correctly on Teensy, but I admit I haven't tested it in a long time.  I'll check it out.

cl0rm, 2018-04-10:
Thank you!

I have the Teensy++2.0 (ATMEGA 1286) and it does not work. I have tested it again.
The KB disconnects, turns backlight off, but does not appear in the Teensy Loader.
With the button this works fine

dhowland, 2018-04-10:
What keyboard are you using and what version of EasyAVR?

cl0rm, 2018-04-10:
I am using a Gigabyte Aivia Osmium (I reverse engineered the matrix and added a teensy++2.0 myself, because the original controller died).
EasyAVR version is 2.06.02.

By the way if you want I can contribute my config files for this board, but I doubt anyone needs it, because the mod required pretty heavy SMD (de)soldering.

cl0rm, 2018-04-10:
yes, it does include the following lines:
teensy = True
import easykeymap.templates.AT90USB1286_16MHz_COSTAR as firmware

dhowland, 2018-04-10:
Hmmm... can you share the .hex file?

cl0rm, 2018-04-10:
yep, here you go:
[osmium_new.zip](https://github.com/dhowland/EasyAVR/files/1896358/osmium_new.zip)

cl0rm, 2018-04-10:
[osmium.py.zip](https://github.com/dhowland/EasyAVR/files/1896363/osmium.py.zip)
this is the config file, if you need the matrix pins.
I mapped boot to fn-Numlock

dhowland, 2018-04-10:
Whelp, I've looked at everything and it all looks perfect.  I can't explain why it isn't working.  I don't have anything with a Teensy++ 2.0 so I can't independently test it.  Not sure where to go from here.  Maybe I'll think of something in the shower.

dhowland, 2018-04-12:
Last ditch thing to try:  try changing Teensy to False and rebuilding.

cl0rm, 2018-04-12:
I will try that

cl0rm, 2018-04-14:
Today I have tried teensy=False, but the behavior is exactly as before. When I press the key combo, the board disconnects, a few milliseconds later the backlight turns off, but it does not reconnect in bootloader mode. Teensy Loader does not detect it

dhowland, 2018-04-14:
Question: when you use SCANCODE_BOOT, you said it resets, but it isn't recognized by the Teensy loader.  But, is it recognized as a keyboard, or is it just dead?

dhowland, 2018-04-16:
wow, okay that changes things.  I'm going to have to make some test builds.

underpk, 2018-07-15:
I have the exact same problem with Teensy 2.0, Once you press it sounds like it go into boot mode but its not.

dhowland, 2018-07-15:
Yes this one really annoys me.  I have checked and double checked the code against the AT90USB1286 manual.  I don't have a board that uses a Teensy 2.0++, which makes it hard to investigate.  I've taken a few months off, I'll try looking again maybe something will pop out.

Are you available to run some test builds on your hardware?

jrauch, 2018-07-29:
@dhowland I'd be happy to run test builds for you - but I'd be equally happy to send you a Teensy 2.0++ if it'd be helpful.

TobiasWegener, 2019-01-07:
Hi everybody, I found in line 184 ` 2,` in the osmium.py file
is this supposed to be there? This looks strange to me, but maybe I am totally off here.

In general, is it possible to use a Teensy++ 2.0 or is it recommended to stick to the standard Teensy2.0?
I am planning to make a custom board I would love to use EasyAVR, that why I am looking into it.
Thank you very much, for this beautiful software.

underpk, 2019-01-07:
> Hi everybody, I found in line 184 ` 2,` in the osmium.py file
> is this supposed to be there? This looks strange to me, but maybe I am totally off here.
>
> In general, is it possible to use a Teensy++ [...]

You can use it with Teensy++ 2.0 (I using it now) but I'm not sure if the software reset already solved or not.

dhowland, 2020-02-09:
The new release v3.01.11 should fix this.

closed 2020-02-09 by dhowland