dhowland/EasyAVR #83: [TODO] 'Tap dance' function

ChenglongWang, 2018-05-14:
Thanks for the contributions on this wonderful work.
It significantly simplify the customize process for newbies like me.

I found the original QMK framework has a nice feature called ['tap dance'](https://github.com/dhowland/EasyAVR) but did not found it in easyAVR. 
I'm curious about that whether this feature will be scheduled.

dhowland, 2018-05-14:
I found the definition of tap dance [here](https://docs.qmk.fm/features/tap-dance).

EasyAVR does not support this because it requires custom programming.  The trade-off for being "easy" is that EasyAVR can't support custom programming.

dhowland, 2018-06-05:
I never actually stopped thinking about this question.  It's an interesting feature and it would be cool to support it.

Though EasyAVR can't support custom programming, it does support the execution of macros, and macros are a sequence of commands.  Perhaps that presents an opportunity.  I may create a branch to test a "tap dance" that accesses into a macro.