dhowland/EasyAVR #84: Mouse buttons don't work unless cursor movement is occurring

TroyFletcher, 2018-05-17:
When pressing an assigned mouse button, there is no effect.

However, when dragging or clicking while moving the cursor, the mouse button registers.

Reproduce by attempting to send a mouse1 button, then apply a mouse_up/down/etc, and reattempt the mouse1.

dhowland, 2018-05-17:
Thanks for the bug report!  There are so many different features, it's really hard to test everything before I do a build.

I'm pretty sure I know exactly what causes this, it's a 5 minute fix.

dhowland, 2018-05-17:
I have released 3.1.6 to fix this issue.  I haven't actually tested it, please let me know if it doesn't work.

closed 2018-05-17 by dhowland

TroyFletcher, 2018-05-18:
Confirmed working great! Thanks for all the great work!