dhowland/EasyAVR #85: Can't find jvm.dll

fattredd, 2018-05-31:
I'm having some issues writing to my board.
```
Running task "Upload to USB AVR with Flip"

The keyboard should be in bootloader mode prior to programming.
If the bootloader has not been activated then the programmer will
not be able to connect and the process will fail.  Activate the
bootloader by using the BOOT key (if it is programmed) or use the
reset switch on your microcontroller.  If the process fails, make
sure the keyboard is in bootloader mode and then try again.

C:\Program Files (x86)\Atmel\Flip 3.4.7\bin\batchisp.exe -device atmega32u2 -hardware USB -operation onfail abort loadbuffer C:\prog\layout.hex memory FLASH erase F blankcheck program verify start reset 0
Can't load jvm.dll
```
I have Java installed, and have even tried adding jvm.dll to my PATH, but to no avail

dhowland, 2018-05-31:
Hmmm... this appears to be a problem with the Flip install.  Does Flip open correctly if you just launch it from the start menu?

fattredd, 2018-06-01:
Looks like I grabbed the version of Flip without JRE. Thanks!

closed 2018-06-01 by fattredd

dhowland, 2018-06-01:
Cool thanks for following up!  I will add something to the docs.