dhowland/EasyAVR #86: Python error when opening JSON files

MiLeG, 2018-07-09:
I have tried loading a json file generated from [KLE](http://www.keyboard-layout-editor.com) and i am getting a python error like this:
[image]

I have tried downloading a .dat layout and a similair error appears. When i downloaded the v2.06.02 version, the .dat file was loaded perfectly.

dhowland, 2018-07-09:
Hi thanks for helping out the project!  Is it possible to share the file that caused the error?

MiLeG, 2018-07-09:
Sure, I have tried out several JSONs, all with similair or same results.

[JSONs.zip](https://github.com/dhowland/EasyAVR/files/2177137/JSONs.zip)

dhowland, 2018-07-10:
Wait, wait.  Are you trying to open KLE layouts using File->Open ?

That doesn't work.  You can only open EasyAVR save files that way.  To use KLE layouts, you have to use the wizard.

closed 2018-07-10 by dhowland