dhowland/EasyAVR #88: ISO key not working (hash/tilde on iso-uk)

Cantello, 2018-07-30:
Hi all!

As proper QMK programming is currently out of my league, EasyAVR really proved to be easy enough to re-program my tada68 keyboard. However, one key is  not programmable, the one to the lower left of the ISO enter key (#' on iso-de, ~# on iso-uk). Regardless of what key I try to bind there, it will not be reproduced.

Any idea what I could be doing wrong?

dhowland, 2018-07-31:
Hi, thanks for helping with the project!

This is probably one of those corners of the software where bugs hide because of lack of testing. I only own US keyboards.

I will take a look. I doubt it's a big problem. I may need a save file, if you can share one that has the problem.

Cantello, 2018-07-31:
Hi David,

thanks for the repl! Further testing shows that this might not necessarily be a problem with EasyAVR as QMK exhibits the exact same problem. No matter what I try to bind to that specific key, it does not produce any output. Stock tada68 firmware works with this key (not the output I want but at least it works) so the soldering and PCB should be okay.

Here is the json file I used for the keyboard:
[tada68_iso-de_easyavr.json.txt](https://github.com/dhowland/EasyAVR/files/2244229/tada68_iso-de_easyavr.json.txt)

dhowland, 2018-07-31:
I took a look at it an there are no obvious flaws.  The hardware config also matches QMK's source code, so I'm not surprised the bug is present in both.

Try this: create a new board with ANSI layout (or <All Keys>), and bind something to the ANSI backslash key (right above Enter) .  Does that work?

Cantello, 2018-08-07:
Searched some more and apparently it was an error in QMK source regarding how the wiring was done. An update to QMK has been done yesterday and the error has been solved: https://github.com/qmk/qmk_firmware/issues/3480

dhowland, 2018-08-07:
Okay, based on that change, I've updated the Tada config in this project.  Next release will have it.

Thank you for your research work, and for following up!

closed 2018-08-07 by dhowland

Cantello, 2018-08-07:
Thanks for the extremely fast update, seems to be working very fine here!