dhowland/EasyAVR #89: Ubuntu 18.0.1 LTS - Unable to install. ?errors

jonathanadams, 2018-08-09:
Trying to install EasyAVR. I have followed the instructions in the guide but no luck.

Python 3.6.5

Here is the terminal output:

```sudo pip3 install -U /home/jonathan/Downloads/easykeymap-3.1.9.tar.gz 
The directory '/home/jonathan/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/jonathan/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Processing ./Downloads/easykeymap-3.1.9.tar.gz
Collecting wxPython>=4.0.1 (from easykeymap==3.1.9)
  Downloading https://files.pythonhosted.org/packages/dd/31/bd55ab40e406a026a7fda0bb5eb61f466682544ae91ac26267c750f5e618/wxPython-4.0.3.tar.gz (68.5MB)
    100% |████████████████████████████████| 68.5MB 32kB/s 
Collecting PyPubSub (from wxPython>=4.0.1->easykeymap==3.1.9)
  Downloading https://files.pythonhosted.org/packages/ab/9e/3b50915d3346971aaa49074425788598ee4907e67c097e013f1a862bd45c/Pypubsub-4.0.0-py3-none-any.whl (63kB)
    100% |████████████████████████████████| 71kB 3.9MB/s 
Requirement already up-to-date: six in /usr/lib/python3/dist-packages (from wxPython>=4.0.1->easykeymap==3.1.9)
Installing collected packages: PyPubSub, wxPython, easykeymap
  Found existing installation: wxPython 4.0.1
    Not uninstalling wxpython at /usr/lib/python3/dist-packages, outside environment /usr
  Running setup.py install for wxPython ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-3096eqku/wxPython/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-u17x4wl6-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    WARNING: Building this way assumes that all generated files have been
    generated already.  If that is not the case then use build.py directly
    to generate the source and perform the build stage.  You can use
    --skip-build with the bdist_* or install commands to avoid this
    message and the wxWidgets and Phoenix build steps in the future.
    
    "/usr/bin/python3" -u build.py build
    Will build using: "/usr/bin/python3"
    3.6.5 (default, Apr  1 2018, 05:46:30)
    [GCC 7.3.0]
    Python's architecture is 64bit
    cfg.VERSION: 4.0.3
    
    Running command: build
    Running command: build_wx
    wxWidgets build options: ['--wxpython', '--unicode', '--gtk3']
    Configure options: ['--enable-unicode', '--with-gtk=3', '--with-opengl', '--enable-sound', '--enable-graphics_ctx', '--enable-mediactrl', '--enable-display', '--enable-geometry', '--enable-debug_flag', '--enable-optimise', '--disable-debugreport', '--enable-uiactionsim', '--enable-autoidman', '--with-sdl']
    /tmp/pip-build-3096eqku/wxPython/ext/wxWidgets/configure --enable-unicode --with-gtk=3 --with-opengl --enable-sound --enable-graphics_ctx --enable-mediactrl --enable-display --enable-geometry --enable-debug_flag --enable-optimise --disable-debugreport --enable-uiactionsim --enable-autoidman --with-sdl
    checking build system type... x86_64-pc-linux-gnu
    checking host system type... x86_64-pc-linux-gnu
    checking for --disable-gui... no
    checking for --enable-monolithic... no
    checking for --enable-plugins... no
    checking for --without-subdirs... no
    checking for --enable-official_build... no
    checking for --disable-all-features... no
    checking for --enable-universal... no
    checking for --enable-nanox... no
    checking for --enable-gpe... no
    checking for toolkit... gtk
    checking for --with-libpng... yes
    checking for --with-libjpeg... yes
    checking for --with-libtiff... yes
    checking for --without-libjbig... no
    checking for --without-liblzma... no
    checking for --with-libxpm... yes
    checking for --with-libiconv... yes
    checking for --with-libmspack... no
    checking for --without-gtkprint... no
    checking for --with-gnomevfs... no
    checking for --with-libnotify... yes
    checking for --with-hildon... no
    checking for --with-opengl... yes
    checking for --with-dmalloc... no
    checking for --with-sdl... yes
    checking for --with-regex... yes
[... 334 lines cut ...]
    configure: error:
    The development files for GTK+ were not found. For GTK+ 2, please
    ensure that pkg-config is in the path and that gtk+-2.0.pc is
    installed. For GTK+ 1.2 please check that gtk-config is in the path,
    and that the version is 1.2.3 or above. Also check that the
    libraries returned by 'pkg-config gtk+-2.0 --libs' or 'gtk-config
    --libs' are in the LD_LIBRARY_PATH or equivalent.
    
    Error running configure
    ERROR: failed building wxWidgets
    Traceback (most recent call last):
      File "build.py", line 1321, in cmd_build_wx
        wxbuild.main(wxDir(), build_options)
      File "/tmp/pip-build-3096eqku/wxPython/buildtools/build_wxwidgets.py", line 375, in main
        "Error running configure")
      File "/tmp/pip-build-3096eqku/wxPython/buildtools/build_wxwidgets.py", line 85, in exitIfError
        raise builder.BuildError(msg)
    buildtools.builder.BuildError: Error running configure
    Finished command: build_wx (0m2.390s)
    Finished command: build (0m2.390s)
    Command '"/usr/bin/python3" -u build.py build' failed with exit code 1.
    
    ----------------------------------------
  Can't rollback wxPython, nothing uninstalled.
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-3096eqku/wxPython/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-u17x4wl6-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-3096eqku/wxPython/```

dhowland, 2018-08-10:
Hi, thanks for reporting!  I don't have an Ubuntu Bionic machine, so I rely on user reports to find issues.

Your system is trying to compile wxPython and it is failing because you don't have the development libraries installed.

Compiling wxPython shouldn't be necessary on Ubuntu Bionic.  What commands are you running?  Are you using a virtualenv?

jonathanadams, 2018-08-10:
Thanks for getting back to me so quick.

I don’t have a virtualenv but I can try if needed?

I first tried to run:

sudo apt install python3 python3-pip python3-wxgtk4.0
sudo apt install dfu-programmer

Then the install command but with no luck. I then tried to add the following:

sudo apt install libwxgtk3.0
sudo apt install libsdl1.2debian
sudo pip3 install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-16.04 wxPython

But also had no success.

jonathanadams, 2018-08-10:
Just noticed the error. I may need to try installing the wxPython GTK3 for 18.04 instead?

https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-18.04/

jonathanadams, 2018-08-10:
Got it working by running:

```sudo pip3 install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-18.04 wxPython```

closed 2018-09-02 by jonathanadams