dhowland/EasyAVR #90: Automod with Left mods

TroyFletcher, 2018-08-28:
Is there a way to use a normal key (S) with a left, rather than a right automod?

I'm running into an issue where the right alt on the system triggers as an Alt Graph, so it triggers a special character rather than a standard Alt. It would be preferable to use Left_Alt + S

Is this a feature that can be added, or is there a way to accomplish this manually on the current version?

Thanks so much, keep up the stellar work!

dhowland, 2018-08-29:
Yeah, this has been asked for a few times.  The reason it is this way is because I have a very limited amount of memory, and the firmware has to fit the automod information into 4 bits.  that means I had to pick one or the other.  Believe it or not, it used to be the left mods, but then someone pointed out that they couldn't do any of their AltGr characters with the left mods, so I changed it to right.

I am going to have to think on this one.  I'm sure there is a way to do it, and it sounds like there is a need.

TroyFletcher, 2018-08-29:
Understood. It's not really a hard requirement, but it would certainly be nice to have.

I have no issue with you closing this if you choose.

Thanks again for this awesome software!

RuanoK, 2020-02-29:
I have the solution for everyone.
Only changing R_Alt to L_Alt is the need.
First I want to provide important information or clarification:

L_Alt = Alt
**R_Alt = Ctrl + Alt = AltGr**

Then if the Automods R_Alt is converted to L_Alt we already have Alt.
And to get AltGr it is enough to activate R_Ctrl + L_Alt.

Only for aesthetic reasons in the end it is good to put them all "Left".

On the web help it is good to explain that Ctrl + Alt = AltGr and that you have to select 2 Automods selections for this case.

It is the solution to get Alt, AltGr and not to create a switch.

dhowland, 2020-03-02:
I have implemented this feature.  A few more days of testing and it will show up in the repo.

dhowland, 2020-03-04:
tested, seems to be working

closed 2020-03-04 by dhowland