dhowland/EasyAVR #92: [Tada68] .bin file doesn't work when flashed according to instructions

danielksato, 2019-03-06:
The .bin file EasyAVR generates for my Tada68 renders my keyboard unusable. I can put the keyboard in flash mode, and copy the file over per the instructions, but when I press Esc, the lights on the bottom turn solid and the keyboard becomes unresponsive to input.

It does not affect the bootloader; I'm able to put the board back in flash mode and successfully reflash both the stock firmware and a firmware image generated from the official site.

Could this be a dependency issue? I'm on Arch Linux running Python 3.7.2.

dhowland, 2019-03-07:
Short answer is I don't know.  I don't have that board and I don't really know anything about its bootloader.  Sorry :/

danielksato, 2019-03-07:
Fair enough - thanks for all the work you've put in on this project! Please feel free to close this issue, if you don't anticipate any other contributors having the chance to dive in.