dhowland/EasyAVR #95: Feature request: KEY WILDCARD.

RuanoK, 2020-03-02:
New "KEY WILDCARD" scan code selector with upper / lower case options in ALT (0123) / ALT (0124) format as input format, ANSI or UNICODE preferably.
Unlimited, able to fill the entire keyboard if desired.
And expose the 2 printable values (not the code) in the LAYOUT.

Reasons to justify the request:
Special keys such as ">,. / *% $ & * (! @ # Etc. are very different in each of the countries, not coinciding with standard English, and the most important thing is that it is a door open to all letters in different languages also if UNICODE will be used.

dhowland, 2020-03-03:
Language localization has been discussed several times before on the forums.  It would be a large overhaul.  I can't do it.

The current software is based on the USB HID standard scancodes, which are named after the English layout.  The keyboard itself knows nothing about Unicode or ASCII or anything other text encodings.  It only knows scancodes, and they don't change for foreign language layouts.

RuanoK, 2020-03-03:
I understand that change is hard work.
Do not think that I want to press you with the following question:

Is it very difficult to make macros that have the "shift" option?
example:
I click on M3 = M3
I click on Shift + M3 = M5 (M5 is any other user-defined macro, associated with M3 for uppercase)

dhowland, 2020-03-03:
That can be done with layers.  Make your shift key a layer selector, and check the same shift key in the "with mods" list.

RuanoK, 2020-03-03:
Thank you.

This is the best software for keyboards, best user friendly.
Congratulations.

closed 2020-03-04 by dhowland