dhowland/EasyAVR #97: Issue using Shift with the new feature Sticky

RuanoK, 2020-03-06:
It is a problem due to the way you have implemented the Sticky feature.
You have coded to make it work like this:

(shift sticky featured)
click shift key
scan code shift down
...
...
(time)
...
...
click key letter
scan code key letter down
scan code shift up
scan code key letter up

if I dont click a letter key there are troublesome if I go to other task as opening programs from task manager, auto opening instances of same app.
And with mouse another problem auto selecting when change the pointer.

I think the process should be:

click shift key
scan code shift down
scan code shift up
...
...
...
(time)
...
...
click key letter
scan code shift down
scan code key letter down
scan code shift up
scan code key letter up

dhowland, 2020-03-06:
The former is definitely the way I intended for it to work.

RuanoK, 2020-03-06:
I know it's not a bug, it's your decision, use it for a while and you'll find the trouble.
Please do not think that I write to bother you. It was just wanting to polish this magnificent software.

Using AutoHotKey script:
-----------------------
(normal shift EasyAvr)
$ shift ::
Input, OutputVar, L1
Send +% OutputVar%
return
---------------------

closed 2020-03-07 by RuanoK