dhowland/EasyAVR #98: Shift Sticky bug, more than a capital letter.

RuanoK, 2020-04-05:
Using sticky key feature, if I write fast I can write 2 capitals letters,
TH as example ,using the bug.
in fast way but as regular typist,
I have experimented with more speed and get 3 and 4 letters capital letters, but this case is forced.
Best regards, Ongoe Allen.

dhowland, 2020-04-05:
This is because the software clears the sticky key on the alpha_up action.  So, if you press two keys at a time they they can both get the sticky modifier.  This was done so that you could hold a key if desired, but the side effect probably isn't acceptable.  I'll look into it.

dhowland, 2020-04-06:
I made a test build that changed this behavior, but it doesn't work.  When the mod is unstuck by the the alpha key downstroke, the mod goes up before the alpha goes down.  This is due to the way the USB HID drivers work.  Fixing this is going to be non-trivial.