dhowland/EasyAVR #100: Edit Menu Error

drolph, 2020-06-04:
When I click on key instead of the edit menu pop up I get an `wxPython: stdout/stderr` pop up with:

> Traceback (most recent call last):
>  File "/Library/Python/3.7/site-packages/easykeymap/gui/layoutpanel.py", line 259, in OnClicked
>   map_dialog.switch_key(self)
>  File "/Library/Python/3.7/site-packages/easykeymap/gui/mapdialog.py", line 130, in switch_key
>    self._enable_tap()
>  File "/Library/Python/3.7/site-packages/easykeymap/gui/mapdialog.py", line 119, in _enable_tap
>    if self.pg.GetPropertyValue('mode') == key_mode_map['Tap Key']:
>TypeError: Unexpected type ("longlong") in wxVariant.

Here's the console:
>UpdateRecents: received results from HIS_XPC_GetApplicationPolicyForURLs
>UpdateRecents: ignoring results because menu isn't open

Checking wxPython version:
> wx.version()
> '4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4'

dhowland, 2020-06-05:
Hmmm...

This might be an issue with WX on OSX.  I don't get this error on my Windows install, and I don't have an Mac to test.

It appears something similar has been reported to the WX people: https://github.com/wxWidgets/Phoenix/issues/1299

If this is indeed an issue, I'm not sure how I can work around it.  I'll play around to see what I can find out.

dhowland, 2020-06-05:
Ok I just upgraded my Windows system to Python 3.8.3 and wxPython 4.1.0 and I get the same error.

```
Traceback (most recent call last):
  File "D:\Code\EasyAVR\keymapper\easykeymap\gui\layoutpanel.py", line 259, in OnClicked
    map_dialog.switch_key(self)
  File "D:\Code\EasyAVR\keymapper\easykeymap\gui\mapdialog.py", line 130, in switch_key
    self._enable_tap()
  File "D:\Code\EasyAVR\keymapper\easykeymap\gui\mapdialog.py", line 119, in _enable_tap
    if self.pg.GetPropertyValue('mode') == key_mode_map['Tap Key']:
TypeError: Unexpected type ("longlong") in wxVariant.
```

so yeah, I think this is a problem with wxPython, not with my code, but I might be able to work around it.

drolph, 2020-06-05:
Thanks for looking into it. I’ll install an older WX module

dhowland, 2020-06-05:
That should work.  I intend to track wxPython releases so I will stick with 4.1.0, which means a workaround until they can fix the issue.  I am going to try and release a new version of EasyAVR tonight after I get the kids to bed.

closed 2020-06-06 by dhowland

dhowland, 2020-06-06:
Ok, give 3.03.02 a try, see if that is better (if you want).

Thank you for the helpful bug report!

drolph, 2020-06-08:
Works now, thanks!

dhowland, 2020-06-08:
That's good.  Also, there is already a pull request to fix the issue in wxPython, so that project should be better for it, too!

commit 07493c0f (2020-06-06): Fix crash when using wxPython 4.1.0

Fixes #100