dhowland/EasyAVR #103: wxPython: stdout/stderr when selecting "Define Keyboard"

nicohofdtz, 2021-09-10:
I used a fresh download of EasyAVR.

Here's the full error message that appeared when I wanted to define a new keyboard:
```
Traceback (most recent call last):
  File "D:\Code\EasyAVR\keymapper\easykeymap\gui\mainframe.py", line 315, in OnNewboard
  File "D:\Code\EasyAVR\keymapper\easykeymap\gui\newboardwizard.py", line 42, in nbwizard
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```

dhowland, 2021-09-10:
Hi thanks for the bug report!

Please provide the following info:
Windows version
Python version
wxPython version
EasyAVR version

I just tested the new board wizard on my Windows 10 PC using the 3.4.1 prebuilt windows binary and it worked without error.

nicohofdtz, 2021-09-10:
Hi,

> Windows version -> 21H1
> Python version -> 3.9.7
> wxPython version -> The User's Guide doesn't say I would need to install wxPython for the prebuilt windows binary.
> EasyAVR version -> 3.04.01

dhowland, 2021-09-10:
Hmmm... ok so this is the same setup I just tried on my own PC and I don't have any problems.

So, looking at the error message, it has something to do with Windows locale.  I'm using US English.  What are you using?

The specific line producing the error is just the following:

    bmp = wx.Bitmap(get_pkg_path("res/wizard.png"), wx.BITMAP_TYPE_PNG)

Which doesn't have any locale information and is correct as far as I can see.

A quick look at the wxWidgets source doesn't provide any obvious hints.  EasyAVR is not internationalized, so I don't touch any of the locale stuff.  It should all be default.  I don't know what the Bitmap module us using it for.

dhowland, 2021-09-10:
Try this:

https://www.dropbox.com/s/sq5qpsp57advo98/easykeymap_windows_3_04_02_test.zip?dl=0

This simply sets the WX Local to English.  It might solve the problem.

dhowland, 2021-09-10:
Actually this seems to be related to wxWidgets/Phoenix#769

commit 3f63386c (2021-09-11): Set locale to US English

Should fix #103
closed 2021-09-11 by dhowland