dhowland/EasyAVR #105: This program can't start beacuse api-ms-win-core-path-l1-1-1.dll is missing

silitekmodder, 2021-11-15:
When trying to start easykeymap the error message "This program can't start because api-ms-win-core-path-l1-1-1.dll is missing" appears. I am using Windows 7 64 bit and tried on two different machines with the same error.

dhowland, 2021-11-15:
Thanks I'll look into it.

dhowland, 2021-11-16:
The "compiled" version of EasyAVR for Windows is now based on Python 3.9.  This version is pretty new, but 3.10 is actually already released, so it's not bleeding edge.  3.9 does not support Windows 7 or earlier.  You have two options:

1. Install Python 3.8 on your system and use the EasyAVR multiplatform release.  (i.e. run from source)
2. Try this, which I found on the web: https://support.microsoft.com/en-us/topic/update-for-universal-c-runtime-in-windows-c0514201-7fe6-95a3-b0a5-287930f3560c

Thanks for the issue!

dhowland, 2021-11-16:
Adding WontFix label because I don't intend to hold steady on older Python releases.  I will update to Python 3.10 when it is a little more mature (after 3.10.2, for example)

closed 2021-11-16 by dhowland