dhowland/EasyAVR #106: Not sure if this is an "issue".  More so a question....  Would this software work with these products?

stevenwcronin, 2021-11-21:
Ecarke 3D Printing Mini USB-Micro Interface 6-Key One-Handed Mechanical Gaming Keypad OSU HID Programming Macro Keys Blue Switches   --> https://www.amazon.com/gp/product/B096Z5B74G

Ecarke USB Mini 3-Key Keypad Mechanical Gaming Keyboard Programming Macro with Software  --> https://www.amazon.com/dp/B08P1GY3GN

These work great, but require some strange software I'd rather not use if I could use EasyAVR with them?   
I'd appreciate any comments,  and I'm willing to try to get it to work, but might need a little help?  

Thoughts?

Thanks!
Steve

dhowland, 2021-11-21:
After a quick search I can't find any hardware specifications online, so I can't be sure.  However my guess is... probably not.

EasyAVR needs a USB-integrated 8-bit AVR microcontroller, with a compatible bootloader.  I don't see evidence that those Ecarke boards meet this criteria.

However, this is a similar product that is definitely supported: https://techkeys.us/products/sixkeyboard 

For what it's worth I also would not install that sketch software on my PC.

closed 2021-11-21 by dhowland

stevenwcronin, 2021-11-22:
Ok.  Thanks for your comments.  I'm reading about 127.0.0.1:7296/ which is the port it's using, etc.  Is there anything you see in the source code that looks specifically concerning?  It seems to work fine, but I can see why you'd be a bit skeptical.