dhowland/EasyAVR #108: Keyboard Macros

keyboardbuldier, 2022-02-05:
When I try to write a macro with the unusual character ä, the program returns an error when I try to build and reprogram.

closed 2022-02-05 by keyboardbuldier

dhowland, 2022-02-05:
Yes I'm afraid the project has no support for internationalization support.  I'm in the US and I basically programmed a 1:1 mapping of ascii codes to scancodes.  It doesn't understand unicode or other keyboard layouts.

That being said, there may be a relatively simple fix for this by using hex codes.  If you post the error message, I can take a look.

keyboardbuldier, 2022-02-05:
Same isue here "unicode macro support"

dhowland, 2022-02-05:
Oh yeah, #35.  I really should do something about that.  At a minimum it shouldn't crash when unsupported characters are entered.